openmv/openmv #2652: New way to prevent I2C fast mode

timdechant, 2025-04-09:
STM32 HAL defines `I2C_FASTMODEPLUS_I2Cx` even when Fast Mode is not supported:
https://github.com/openmv/openmv/blob/22cbc5d6bbc286f23b0ea98fc66daf79190e27a4/src/hal/stm32/f7/include/stm32f7xx_hal_i2c_ex.h#L73-L77

...which causes an error here:
https://github.com/openmv/openmv/blob/22cbc5d6bbc286f23b0ea98fc66daf79190e27a4/src/omv/ports/stm32/omv_i2c.c#L190-L194

The macro `IS_I2C_FASTMODEPLUS()` or something similar might be helpful here, except that the HAL includes a cast in defining `I2C_FASTMODEPLUS_I2Cx`.  So the preprocessor cannot handle this in any directive!

I suggest adding an option like `OMV_I2C_DISABLE_FASTMODE` which can be defined to entirely bypass this code.

iabdalkader, 2025-04-09:
> ...which causes an error here:

I think this is correct. Don't you need FMP to achieve the highest speed? If it's not supported it should fail. Note that nothing actually uses `OMV_I2C_SPEED_FAST` except for MLX90x FIR sensors.

timdechant, 2025-04-09:
It fails to compile.

iabdalkader, 2025-04-09:
> It fails to compile, so there should be a compile-time method to disable it.

How does it fail to compile? Can you post the error or let me know how to reproduce it? It's very easy to add options, just need to be sure we really need it first.

timdechant, 2025-04-09:
Right, I skipped a step.

`HAL_I2CEx_EnableFastModePlus()` is called always, but is only defined if FastMode is supported:
https://github.com/openmv/openmv/blob/22cbc5d6bbc286f23b0ea98fc66daf79190e27a4/src/hal/stm32/f7/include/stm32f7xx_hal_i2c_ex.h#L116-L119

iabdalkader, 2025-04-09:
> `HAL_I2CEx_EnableFastModePlus()` is called always, but is only defined if FastMode is supported:

Same as `IS_I2C_FASTMODEPLUS` (at least for F7, H7 and N6). What if we just test for that instead?

timdechant, 2025-04-09:
`#if IS_I2C_FASTMODEPLUS(I2C_FASTMODEPLUS_I2Cx)` fails because it includes a cast.

iabdalkader, 2025-04-09:
No, the macro itself is only defined if FMP is supported. I checked F3, F7, H7 and N6 (always supported). Unless I missed something, this should work. Left a comment on your PR.

timdechant, 2025-04-09:
Oh!  I thought I tried that; I will recheck.

iabdalkader, 2025-04-09:
Fixed.

closed 2025-04-09 by iabdalkader

timdechant, 2025-04-09:
> If it's not supported it should fail

Should I add an `assert_param` to check for attempted FMP when not supported?