openmv/openmv #2665: No bluetooth connection on latest firmware update

sSpectrals, 2025-04-19:
**Describe the bug**
A clear and concise description of what the bug is.
Cannot detect a connection with Bluetooth between camera-camera or phone-camera
(Tested with RT1062)

**how it was fixed** 
I saw a post on the forum that said they had the same issue, they downgraded and it worked.
I did the same thing. I downgraded to 4.5.5.
In the same forum post I believe it said OpenMV fixed the bug, but it might have accidentally reappeared in a new firmware update. 
[https://forums.openmv.io/t/ble-in-rt1062-pairing-error/9291](url)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to examples and open Bluetooth 
2. Run Code
3. Try to find a connection with an app on your phone or another Bluetooth module
4. Cannot find the camera

**Expected behavior**
The user should be able to find a camera to connect to.

**Please complete the following information**
 - IDE version: 4.5.0
 - Firmware version: v4.6.20
 - Board: OPENMV RT1062
 - OS: Windows 11

kwagyeman, 2025-04-19:
For the RT1062 please install the latest dev release via Tools->Install Latest Dev release. This will fix the issue.

iabdalkader, 2025-04-19:
If you install the dev firmware, you won't be able to use built-in models. ROMFS support is not ready yet.

closed 2025-06-06 by kwagyeman