openmv/openmv #2674: sensors/genx: Add ioctl to return raw event data?

iabdalkader, 2025-04-24:
Should we? Given the interest in this.

kwagyeman, 2025-04-24:
Yeah, I can send a PR just to make a raw events available and add an ioctl or something so you can get them from the image.

iabdalkader, 2025-04-24:
You can prealloc an ndarray or a buffer, and read events into it? Anyway, this needs to be finalized first before any csi/sensor changes https://github.com/openmv/openmv/pull/2669

kwagyeman, 2025-04-24:
Yeah, an ndarray would be the way to go here. Would have to experiment with what is the fastest thing to do.

kwagyeman, 2025-05-27:
I will make the API work as following:
 
1.	You will be able to set a frame size other than B320X320 which will allow you to change the event buffer size. This will only be enabled after setting an event mode ioctl.
1.	The frame size set here defines the snapshot res too.
2.	We’ll provide an ioctl that will populate a preallocated 2d uint16 ndarray with event data.
3.	If the user wants an image they call also call snapshot(), this will generate an image at the selected resolution, but, have quite a bit of CPU overhead.
1.	Folks will want to not call snapshot() once they deploy their app.
 
From the driver level side of things, our frame buffer architecture is just receiving a buffer of bytes in the background always. Calling snapshot() post-processes that buffer into an image and provides a frame. Or, calling the new icotl we’ll add post-process the buffer fills an ndarray. I’ll make the driver such that you can call either in any order. Note though that calling one or the other will eat the event buffer.
 
The preallocated ndarray will have rows equal the size of the event buffer. The columns will be [type, ts_s, ts_ms, ts_us, x, y]. This buffer is meant for the EVT2.0 format. Because we just populated a pre-allocated array, the user can keep one event buffer or make a new one per frame—their choice.
 
Type will be an enum for type (CD_LOW, CD_HIGH, PXRSTN_TRIGGER_RISING, EXT_TRIGGER_RISING, PXRSTN_TRIGGER_FALLING, EXT_TRIGGER_FALLING).
 
The timestamp will be 3 uint16 values. ts_s will be the seconds, ts_ms will be the milliseconds, ts_us will be the microseconds. This fits your 34 bit timestamp with 1us resolution, but, makes it easy to parse. I will automatically, post-process the evt_time_high event and combine that with the low-timestamp bits in events so that this doesn’t need to be done at the python level.
 
Given I will post-process the evt_time_high event the number of the events in the ndarray will not necessarily match the size of the event buffer. Empty spaces at the end will be filled with NULL events. Also, EVT 2.0 OTHER and CONTINUED events will be tossed. Let me know if these should be kept.
 
Finally, we’ll add a function that can turn an event ndarray back into an image for viewing. This can be used to avoid snapshot() completely and/or for viewing recorded data and etc.

iabdalkader, 2025-05-27:
> The terminology though is ugly here though, maybe adding another ioctl to set set_pixformat/framesize would make it nicer looking:

It's not, all scripts start by doing that. If you set the default resolution, that just gives you the max possible number of events, right? Why make the buffer any smaller/bigger?

Also, it's possible to set default framesize/pixelformat in init, or from the ioctl if they're not set.

kwagyeman, 2025-05-27:
> If you set the default resolution, that just gives you the max possible number of events, right? Why make the buffer any smaller/bigger?

It's just the speed at which you get buffer returns. Prophesee's STM32F7 demo kit sets the buffer size at 2048 events. The B320X320 resolution would be 25600 events. In low activity situations, this means you are waiting a long time for the buffer to fill up.

With a 2048 event buffer, you're looking at like 1000+ buffer returns a second when lots of things move in the image.

iabdalkader, 2025-05-27:
I see.. Can we calculate the closest framesize from the buffer size? And round down? This way you could set it in the ioctl (if not set).

kwagyeman, 2025-05-27:
Yeah, that's doable. This works nicely, as then you're not specifying the buffer size indirectly from the frame resolution, which is awkward. And then the icctl returns the number of collected events.

```
sensor.reset()
evt_buffer = np.zeros((2048, 6), dtype=np.uint16)

while True:
   evt_count = sensor.ioctl(sensor.IOCTL_GENX_READ_EVENT, evt_buffer)
   #.....
   img = image.Image(evt_buffer)
```

iabdalkader, 2025-05-27:
>    evt_count = sensor.ioctl(sensor.IOCTL_GENX_READ_EVENT, evt_buffer)

Good idea.

closed 2025-08-09 by iabdalkader