openmv/openmv #2676: Tthoughts on dual CSI support.

iabdalkader, 2025-04-28:
This is not meant to support two actual main sensors, I don't know if it can. It's intended to support an auxiliary or secondary sensor with a different interface (like the Lepton).

Possible solutions:

1- Duplicate most of the csi struct. To avoid duplicating `i2c_bus` and port bits (such as dma, dcmi etc..), which will be unset/unused for `csi1`, use pointers set to common static instances when initialized.

```C
// Port bits (such as dcmi, mdma etc..).
typedef struct _omv_csi_priv {
#ifdef OMV_CSI_PORT_BITS
    OMV_CSI_PORT_BITS
#endif
} omv_csi_priv_t;

static omv_csi_priv_t omv_csi_priv;

typedef struct _omv_csi_dev {
    bool primary;     // Not really needed could test priv==NULL

    uint32_t chip_id;
    uint8_t slv_addr;

    omv_i2c_t *i2c_bus;

    omv_csi_priv_t *priv;  // Can be NULL for csi1.

    // Sensor function pointers
    int (*reset) (omv_csi_t *csi);
    int (*sleep) (omv_csi_t *csi, int enable);
    int (*snapshot) (omv_csi_t *csi, image_t *image, uint32_t flags);
    ....
} omv_csi_t;

omv_csi_t csi[OMV_CSI_DEV_COUNT] = { 0 };

void omv_csi_init_default() {
    for (i=0; i<csi_count; i++) {
        csi[i].i2c_bus = &csi_i2c;
    }
    csi[0].priv = &omv_csi_priv;
}
```

This can be used like:

```C
for (i=0; i<csi_count; i++) {
    csi[i].fb = framebuffer_get(i);
    csi[i].snapshot = omv_csi_snapshot; // will be changed for dev[1] in lepton for example.
}

void DCMI_IRQHandler(void) {
    HAL_DCMI_IRQHandler(&csi[0].priv->dcmi);    // or csi[0].dcmi if duplicated.
}

// id passed from `py_csi_sleep(id=0)`
__weak int omv_csi_sleep(size_t id, int enable) {
    omv_csi_abort(&csi[id], true, false);
    if (csi[id].sleep != NULL && csi[id].sleep(&csi[id], enable) != 0) {
        return OMV_CSI_ERROR_CTL_FAILED;
    }
    return 0;
}
```

==========================================

2- Using a csi struct and sub-devices:

```C
typedef struct _omv_csi_dev {
    bool primary;               // or id == OMV_CSI_ID_PRIMARY

    uint32_t chip_id;
    uint8_t slv_addr;

    omv_i2c_t *i2c_bus;
    omv_csi_t *csi;              // could avoid this with containerof.

    // Sensor function pointers
    int (*reset) (omv_csi_t *csi);
    int (*sleep) (omv_csi_t *csi, int enable);
    int (*snapshot) (omv_csi_t *csi, image_t *image, uint32_t flags);
    ....
} omv_csi_dev_t;

typedef struct _omv_csi {
    bool reset_polarity;
    bool pwdn_polarity;
    
    #ifdef OMV_CSI_PORT_BITS
    OMV_CSI_PORT_BITS
    #endif

    omv_csi_dev_t dev[2];
} omv_csi_t;
```

Can be used as follows:

```C
for (i=0; i<array_size(csi); i++) {
    csi.dev[i].fb = framebuffer_get(i);
    csi.dev[i].snapshot = omv_csi_snapshot; // will be changed for dev[1] in lepton for example.
}

void DCMI_IRQHandler(void) {
    HAL_DCMI_IRQHandler(&csi.dcmi);
}

// id passed from `py_csi_sleep(id=0)`
__weak int omv_csi_sleep(size_t id, int enable) {
    omv_csi_abort(&csi.dev[id], true, false);
    if (csi.dev[id].sleep != NULL && csi.dev[id].sleep(&csi.dev[id], enable) != 0) {
        return OMV_CSI_ERROR_CTL_FAILED;
    }
    return 0;
}

//  Every single function will need to be changed to accept an `omv_csi_dev_t *`:

static int read_reg(omv_csi_dev_t *dev, size_t id, uint16_t reg) {
    uint8_t reg_data;

    // To get parent struct (if needed for any reason), use `containerof`:
    omv_csi_t *csi = (omv_csi_t *) containerof(dev);
    // or dev->csi.

    if (omv_i2c_readb(csi->i2c_bus, dev->slv_addr, reg, &reg_data) != 0) {
        return -1;
    }
    return reg_data;
}
```

==========================================

Both require many changes to `common/omv_csi.*` and `modules/py_csi.*` and `ports/omv_csi.*`

1- Requires way less changes, but some common resource (such as reset, power polarities) will still be duplicated. And those will have to be set on one or the other instance, and copied.
2- Nothing will be duplicated, common resources are accessed through the main struct (`csi->x`), but requires changes to every file.

==========================================

Some corner cases/unresolved issues:

* `reset()`, `shutdown()` etc.. use shared power/reset GPIOs. If one csi is reset, the other resets. Maybe some functions need to act on all CSIs? i.e., reset() resets all.
* Set/get clock frequency, also shared.
* I assume rotation will affect both?
* There are probably many more that I missed.

kwagyeman, 2025-04-29:
> reset(), shutdown() etc.. use shared power/reset GPIOs. If one csi is reset, the other resets. Maybe some functions need to act on all CSIs? i.e., reset() resets all.

I think it's fine to gang all the interfaces together for these global functions. This is how the physical hardware modules are anyway.

> Set/get clock frequency, also shared.

Yes, but, we can control the hardware design to limit things to one clock output too. If a second clock of a different freq is needed it's pretty simple to just add a osc for it.

> I assume rotation will affect both?

Hmirror/vflip/transpose could all be shared or separate. From the users point of view, sharing them will be more friendly since the sensors are pointing the same way. The csi driver could iterate through all sensors drivers to apply the settings. Having each separate offers more control, though, and may be better in the event sensors are not all pointing the same way.

---

> This is not meant to support two actual main sensors, I don't know if it can. It's intended to support an auxiliary or secondary sensor with a different interface (like the Lepton).

Yes. After studying what you want to do for a while. I think option 1 is fine to go with.

As for: 

> To avoid duplicating i2c_bus and port bits (such as dma, dcmi etc..), which will be unset/unused for csi1, use pointers set to common static instances when initialized.

Could these not just be global in their own `omv_csi_priv_t` struct which is independently manipulated?

Finally, using pointers in each csi struct to point back to an i2c_bus seems like a good idea to future proof things if multiple I2C buses are ever used. It's not really a huge cost.

iabdalkader, 2025-05-13:
While Option 1 is very doable, and it could allow different driver data for each csi, it will require even more changes, so let's put a pin in it for now. To "minimize" the changes I just duplicated the whole CSI struct and the code that needs to use driver data gets called with the right CSI (the one with those handles initialized). 

Anyway, as I dug deeper into this I realized the Python API I proposed (passing ID to every function) is just not practical and very messy to implement. You'd be basically passing "self" so why not make it a class? So I started working on a new modern API that enables more than one sensor. The following works now:

```Python
import csi

csi0 = csi.CSI()
csi0.reset(hard=True)
csi0.pixformat(csi.RGB565)
csi0.framesize(csi.QVGA)
csi0.snapshot(time=100)

csi1 = csi.CSI(cid=csi.SOFTCSI)
csi1.reset(hard=False)
csi1.pixformat(csi.RGB565)
csi1.framesize(csi.QVGA)
csi0.snapshot(time=100)
```

A few notes/issues:

* The new Python API can co-exist with the old "sensor" module (both can be enabled at the same time) until the old one becomes obsolete and gets removed eventually. 

* The sensors can be detected in any order, and you can always get the "main" one even if it's at any index (via `omv_csi_get(-1)`)  or lookup a specific one via chip ID. However, one drawback to this approach is that FB sizes should be the same, otherwise the main CSI may get the smaller FB.

* I had to make more functions instance-specific (via pointers), such as abort(). This allows the sub-csi to set it to NULL or override.

* Code is littered with `framebuffer_update_jpeg_buffer` which have to be removed, otherwise JPEG FB oscillates between FB0 and FB1. This is called before changing formate, framesize etc.. to flush the last frame, I'm not sure this is actually needed.

* Related to the above, only one sensor should update JPEG buffer, so I added a new arg to snapshot (`snapshot(update=True)`). By default it does the old behavior, note how it's used in the example for csi1.

* Two approaches to reset/shutdown:
1- Support hard/soft reset. If hard=False (default=True) only the state is reset, GPIO pins aren't toggled. This is the current appraoch and it works well.

2- Make reset/shutdown module-level functions that operate on all sensors in a loop.

[image]

Finally, I'll push a new branch based on N6 for testing later.

kwagyeman, 2025-05-13:
> The sensors can be detected in any order, and you can always get the "main" one regardless of its position in the array, via omv_csi_get(-1), or lookup a specific one via chip ID. However, one drawback to this approach is that FB sizes should be the same, otherwise the main CSI may get the smaller FB.

We're going to be supporting 5MP camera sensors with much lower res secondary sensors. I think it's important to make the logic here smarter to avoid needing the FB's to be the same size. 

> Code is littered with framebuffer_update_jpeg_buffer which have to be removed, otherwise JPEG FB oscillates between FB0 and FB1. This is called before changing format, frame size etc.. to flush the last frame, I'm not sure why this is needed, I think it's safe to remove them?

Not really, otherwise, the last frame won't be flushed. I think adding the update=True flag to each one should be done. As for the secondary secondary sensor... maybe it's update=True flag defaults to False since it's not the primary? Only the primary's defaults to True?

> Related to the above, only one sensor should update JPEG buffer in snapshot, so I added a new arg to (snapshot(update=True)). By default it does the old behavior, note how it's used in the example for csi1.

See above thought.

> Two approaches to reset/shutdown:
1- Support hard/soft reset. If hard=False (default=True) only the state is reset, GPIO pins aren't toggled. This is the current appraoch and it works well. 2- Make reset/shutdown module-level functions that operate on all sensors in a loop.

I like option 1, as for the default, maybe also use the sensor being the primary to set the default to True and not primary to set the default to False.

---

Overall, this is an excellent idea!

kwagyeman, 2025-05-13:
> But why do we need the last frame flushed after say changing the framesize or pixel format? For example, here: https://github.com/openmv/openmv/blob/master/common/omv_csi.c#L694
Also, if for some reason you want to capture an image, switch res/format and then capture another image, could you just flush the buffer explicitly?

Folks don't know they need to do this explicitly since you typically don't have to flush buffers. However, a minority of folks will dynamically change the resolution... so, for the ones who do this they can be educated to need to do an explicit flush.

However, the current API generally tries to avoid the user having to flush frame buffers themselves. Depends on if you want to keep that behavior or not. I don't forsee removing this behavior causing too much of an issue though. Most folks use the default script format of setting the pixformat/res and hever changing it.

> These flags are set to defaults that assume a single sensor (common case) and to maintain the old behavior you'd expect. For the special dual csi case, let the user decide which sensor grabs the JPEG buffer by overriding the defaults.

I guess you don't know which sensor the user wants. So, it would be a surprise if they tried to use the secondary sensor and it required modifying the default from False to True to get an image out.

iabdalkader, 2025-05-14:
> Yeah, I guess you don't know which sensor the user wants. So, it would be a surprise if they tried to use the secondary sensor and it required modifying the default from False to True to get an image out.

Exactly. Also, note that the API allows you to use both, or either one individually. So if you use CSI1 and the default is false, there won’t be any streaming.

iabdalkader, 2025-07-20:
Fully implemented.

closed 2025-07-20 by iabdalkader