openmv/openmv #2769: Zephyr integration protocol

josuah, 2025-07-17:
Hello to the OpenMV team and community!

I have been thinking for some time about how to best take advantage of OpenMV cameras over a wired link in various scenarios:

Extending the protocol and driver support abilities:

- `OpenMV Cam` <--I2C/SPI/QSPI/SDIO/UVC--> `Zephyr MCU` <--CAN/EtherCAT/Modbus/WiFi/LTE--> `...`

Aggregation at the scale of a large many-camera deployment:

- `OpenMV Cam` [x30] <--RS232--> `Zephyr MCU` [x5] <--USB--> `Linux MPU`

Example of effort integrating OpenMV cameras in the context of drones, which you might have heard about:

- https://store.3dr.com/3dr-quad-zero-pro/

The final goal being improving Zephyr support for OpenMV to allow OpenMV camera to declare their feature without having to modify the Zephyr side, which only acts as a stateless protocol gateway.

To give a practical example:

```
OpemMV: I am a sensor device, I detect several things:
[0x1253]: "human presence", (q15_t)score, (uint16_t[4])bounding_box, +EVENTS +BURST +SNAPSHOT
[0x1254]: "face detection", (q15_t)score, (uint16_t[4])bounding_box +EVENTS +SNAPSHOT +ONOFF
[0x2332]: "cuve 3 level", (q15_t)content_level, (uint16_t)total_capacity_liters, +ONOFF

Host: Requesting 0x1254 value 'score'

OpenMV: 0.03

Host: Requesting 0x1254 value 'score'

OpenMV: 0.01

Host: Requesting 0x1254 value 'score'

OpenMV: 0.02

Host: Requesting 0x1254 value 'score'

OpenMV: 0.97

Host: Requesting 0x1254 value 'bounding_box'

OpenMV: [36, 75, 235, 274]

Host: Requesting snapshot at coordinates [26, 65, 245, 284]

OpenMV: [0x34, 0x32, 0x33, 0x27, 0x22, ..., 0x67, 0x88]
```

Such protocols could be very transparent to work with thanks to Python, and have a dedicated Zephyr + Linux driver, along with mappings over a plethora of protocols. Then no need to upgrade the Zephyr part when the OpenMV part is tweaked.

Then various software can be made "OpenMV compatible" as in "I support any OpenMV camera whatever it detects" (wink wink [Golioth](https://blog.golioth.io/tinymcp-unlocking-the-physical-world-for-llms-with-mcp-and-microcontrollers/)).

P.S. thanks to @zacck for the help putting this together.

josuah, 2025-07-17:
Another way to describe the problem: the problem originally faced:

[image]

Video stream over various mediums (USB, Ethernet...) is very well covered by protocols (UVC, (S)RTP, SRT...), replacing that by a stream of high-level information (such as events), the problem of arise: this high-level information may move across multiple protocol (i.e I2C -> RS485 -> MQTT -> datacenter) and be integrated on very diverse ecossytems (ROS2, Web USB, CANopen, servers-side applications...).

=> Is there some similar project ongoing so I can join the effort?
=> What should such a protocol ideally look like at a high-level?
=> Should I instead aim for the forum for such kind of discussions?

It is also possible to make it a 3rd-party `mip` package and list it on [`mim`](https://checkmim.com/), none of this forces OpenMV to maintain anything of course!

Thank you!

josuah, 2025-07-17:
> are there no systems that yet do this?

I did not know about SenML, this seems quite a good model! Maybe the solution to split data definitions (data type, semantics) and "mappings" (how to marshall and send it over various protocols).

A very low-level (I2C/I3C) example is MIPI [Camera Command Set (CCS)](https://2384176.fs1.hubspotusercontent-na1.net/hubfs/2384176/Webinars/MIPI-Webinar-Bootstrapping-MIPI-CCS-Software-Support-with-CCS-Tools.pdf) for which there is already a [generic drivers in Linux](https://docs.kernel.org/driver-api/media/drivers/ccs/ccs.html).

I will go survey more and ask around and report here... Thank you for the prompt reply!

beriberikix, 2025-07-17:
Inter-mcu communication is a deep 🐰 🕳️ ! If you go on reddit or various embedded Slacks, everyone has their own favorite framing protocol, schema etc. There's many practical factors to consider too, like resources available, code reuse (ex. do you already have a networking stack,) security requirements, etc.

I'll start with the simplest case; 1:1 MCU, text data. IP is surprisingly common for this. There could be several reasons why you wouldn't like this, CoAP + CBOR + SenML (and over SLIP) can serve a lot of use cases. You might not need all of that for this application but it can check a lot of boxes (even can support encryption) while fitting onto many Zephyr hosts and works in MP.

(yes, I've heard of COBS! And protobufs!)

We've actually done a lot of work with the [SMP protocol](https://docs.zephyrproject.org/latest/services/device_mgmt/smp_protocol.html). Besides being well-supported in Zephyr (there's a Python library as well,) it can be easily extended with other commands. It'll never be as optimal as a purpose-built protocol but it works in a lot of cases.

There's a different protocol I've been tracking called [MCTP](https://en.wikipedia.org/wiki/Management_Component_Transport_Protocol) that recently got basic support in [Zephyr](https://docs.zephyrproject.org/latest/samples/modules/pmci/mctp/mctp.html). It was designed for servers that have lots of MCUs that need to communicate but it doesn't have to be. It also incorporates device discovery (part of Management Controller Host Interface (MCHI)) that could be adapted. Unfortunately the MCTP implementation is pretty basic and definitely doesn't have MCHI implemented. But worth mentioning for future reference!

I think the auto-discovery is interesting but typically don't have the extra resources on an MCU (and know generally what devices will be attached.) You might consider developing these as drivers in Zephyr or consider a subsystem to make them more easy to integrate / configure (of course there's the option to make them run-time configurable!)

That's a bit of stream of thoughts that I hope were helpful. I'll give it more thought as well!

josuah, 2025-07-18:
Some examples:

- [SenML](https://www.iana.org/assignments/senml/senml.xhtml) - selected by IANA
- [One Data Model](https://onedm.org/) - using SenML types [[1]](https://github.com/w3c/wot-profile/issues/29) [[2]](https://github.com/one-data-model/language/blob/master/sdf.md#odmdata) [[examples]](https://github.com/one-data-model/playground/tree/master/sdfObject)
- [ROS data types](https://github.com/rticommunity/ros-data-types/tree/master/sensor_msgs/msg) - low-level types for robotics
- [labelformat.com supported format list](https://labelformat.com/formats/) for computer vision annotations

SenML seems quite widespread and could make some solid basis.

Reusing ROS data types (outside of any ROS software) is tempting as tested against many actual sensors and [cameras](https://github.com/rticommunity/ros-data-types/blob/master/sensor_msgs/msg/CameraInfo.idl) and [compressed](https://github.com/rticommunity/ros-data-types/blob/master/sensor_msgs/msg/CompressedImage.idl) [images](https://github.com/rticommunity/ros-data-types/blob/master/sensor_msgs/msg/Image.idl), and OpenMV-specific types can be proposed. ROS data types can of course be used outside of ROS.

zacck, 2025-08-08:
There exists a `ROS` [module](https://github.com/micro-ROS/micro_ros_zephyr_module) for Zephyr that can serve as a client in communicating with  an external system perhaps a small POC would be to use such a module to enumerate functionalities  and capabilities. 
I also spot a related effort in [here](https://github.com/fabianschilling/openmv_cam) perhaps these can sort of show what has  been tried and enable us to take another stab at it.

zacck, 2025-08-08:
Ah thanks for this, I guess I am looking at the python ecosystem and trying to see what have been done RE: publishing information (OpenMV in this case) and subscribing to information ( Zephyr in our case). Does not have or need to be ROS.

josuah, 2025-08-19:
> This isn't an easy thing to get working across all cameras

I forgot about this aspect, thank you.

> However, our current interface library can already do remote callbacks and etc. You could build the ROS system ontop of it.

Is it [`pyopenmv`](https://github.com/openmv/openmv/blob/master/tools/pyopenmv.py)? Better

> The most realistic USB interface to the board right now as a MicroPython device is via REPL that appears on USB and sending a script to configure the camera and report data via that.

This sounds like something easy to quickly adapt to a lot of use-cases.

Which of the two approach would you recommend trying first?

Thanks again for the guidance.

kwagyeman, 2025-08-27:
@josuah - I was talking to a customer this morning and they needed a high bandwdith data link to connect our cameras together. As we were working through a solution I realized that the ethernet interface is the best thing to use for this.

1. You can wire up systems using proper ethernet magnetics at any distance. And with PoE you get power.
2. You can also wire up the ethernet PHYs using blocking caps to each other without magnetics.
3. You can use switches to have multiple boards at the same time.
4. If there's no DHCP, then you just set a static IP on each board and you can communicate.
5. Best of all, via ethernet and TCP and UDP you can do any protocol. E.g. you can mimick a ROS2 compliant node. Whatever you want. There's no limit.

Given this, I don't really see the value in trying to force the USB interface to be something it isn't when ethernet offers a better solution. On the RT1062 it's 100Mb/s and on the N6 it's 1000Mb/s. So, it's the fastest interface onboard the device. It makes sense to use it instead. Additionally, the Ethernet IP already has a DMA ring buffer system and is capable of actually moving packets at extremely high speeds.

Anyway... so, given the above. I think this answers the discussion question. Which is, how to network the OpenMV Cam. And the answer then is, via Ethernet or Wifi using standard networking protocols. Any zephyr board will support this.

Does this cover what you need?

iabdalkader, 2025-08-27:
To add to that last thought, I'm working on a new protocol for our cameras to replace the old one, prompted by wanting to create a more reliable communication protocol. However, while working on this, I realized I could also make it transport-agnostic. The new protocol introduces the concept of channels, which can be physical (like USB, UART, I2C, or Ethernet) or logical (like the frame buffer, REPL data, or anything, really). Here's the interesting part: it will be possible to implement those channels in MicroPython. This means that anything available to MicroPython can be used to transport protocol data just like USB does right now (well, that is if it works 😅).

As for other hosts, we could provide a reference Python/C protocol implementation that you could just include in any project. It's a bit ambitious, but we could even push a protocol implementation to Zephyr, with support for multiple transports, which should give you the same level of control the IDE has, from any Zephyr host. That is, if that's something they'd accept.

Anyway, this is quite an undertaking, and it's still in the very early phases, so there are many details to work out, but I thought I should let you know where this is going.

josuah, 2025-08-27:
Ethernet avoids all the difficulties of dealing with USB while bringing all the practical advantages. Higher cost, higher distances, perfect! There is not yet a complete support for video over network on Zephyr, but there is definitely [interest for it](https://github.com/zephyrproject-rtos/zephyr/discussions/92843).

> a reference Python/C protocol implementation [...] we could even push a protocol implementation to Zephyr [...] that is if that's something they'd accept

This seems to stand right in-between hardware support and protocol support.
This could be a good fit for the upcoming Gstreamer-for-Zephyr called libMP (MediaPipe, slides of the talk not yet available).

> which should give you the same level of control the IDE has

Reusing the protocol developers use for other purposes sounds very good! Only one protocol to maintain, expectation is if "it works in my laptop", then "it works on my edge LTE gateway".

josuah, 2025-09-07:
Thank you! I will have to wait that a few projects at `$work` are complete to find time for it, but that is definitely promising.

It seems a very nice protocol, as if POSIX APIs all got mapped to a protocol (i.e. poll). Sounds convenient for something low-level between MicroPython and and OS/RTOS.

iabdalkader, 2025-09-13:
> as if POSIX APIs all got mapped to a protocol (i.e. poll

It also supports configurable async events, which are used to send system events like frame-ready, soft-reboot, etc.. or custom channel events.

Here's the protocol running on a Uart transport implemented in Python. It's extremely slow at 115200, but it's working 🤷🏼‍♂️ 

https://github.com/user-attachments/assets/e20996e1-59a4-46d4-8e23-ebd8441eac65