openmv/openmv #2770: I want to expose additional UART on Openmv RT0106 board.

anand2532, 2025-07-18:
I need two UART port support in the Openmv RT0106 board. From the information from the datasheet and pin diagram: 
* its clear that UART0 and UART1 is available on the board.
* UART1 is exposed on the pins and the port UART0 I was not able to access through pins. 
* I want to expose UART3 to P13 and P14 or any other pins. 

I just want to confirm, is it possible to change the firmware and expose UART3. Any additional detail I need to know when changing the firmware.

closed 2025-07-18 by kwagyeman

anand2532, 2025-07-18:
Hey, Is it possible to chage the UART3 to expose on pins or other UART port to expose on pins. I need atleat  2 UART with this module. Can you please confirm if it is possible to expose any other UART or use UART0 for deployment.

anand2532, 2025-07-18:
I have forked the repo, trying out to expose any other UART, if is there any other possibility please let me know. Thanks

* Or is there any possibility to expose UART3 on pins, disconnecting wifi/ble through firmware.
*

kwagyeman, 2025-07-18:
We exposed all the UARTs that are possible to expose. There isn't a second UART option on the I/O pins.

anand2532, 2025-07-18:
Hey @kwagyeman, Can I use the usb c port of the module to connncet to the module via usb to ttl converter. if that is possible, on which port should i access to get the data.