openmv/openmv #2789: The Servo and independent PWM configuration cannot work simultaneously.

underthedom, 2025-07-29:
**Describe the bug**
The servo and the standalone PWM function both use TIM4. When reconfiguring one of them (depending on the order of initialization), the register settings related to the other will be overwritten.
Related code files:
  lib/micropython/ports/stm32/servo.c
  lib/micropython/ports/stm32/timer.c

**To Reproduce**
Copying these codes into the IDE and running them will result in abnormal Servo duty cycle behavior.
#--------------------------------------------------------
```
import time
from pyb import Pin, Timer, Servo

s1 = Servo(1)  # P7
s1.angle(90)

tim = Timer(4, freq=1000) #1KHz
ch2 = tim.channel(2, Timer.PWM, pin=Pin("P8"), pulse_width_percent=50)  # P8

while True:
    pass
```
#--------------------------------------------------------

**Expected behavior**
P7: 2.5ms/50Hz 
P8: 0.5ms/1KHz

**Screenshots**
Abnormal waveform (from the servo) --> P7: 5us/1KHz(Same frequency as P8)

[image]

**Please complete the following information**
 - IDE version: 4.5.1
 - Firmware version: 4.7.0
 - Board: OPENMV-H7 Plus
 - OS: Windows

kwagyeman, 2025-07-29:
Since they use the same timer you can't have widely different functions on the pin. This is a limitation of the hardware.

underthedom, 2025-07-29:
你好，已成功收到你的邮件。

closed 2025-07-29 by kwagyeman