openmv/openmv #2823: Nicla Vision – MJPEG Streaming Example fails with CYW43 WiFi errors

Ssumeetdubas, 2025-08-25:
Description:
On the Arduino Nicla Vision board, the MJPEG streaming example does not work.
The board connects to Wi-Fi successfully and displays an IP address.
However, the webpage at that IP does not load, and no camera stream is visible.
The serial monitor shows continuous CYW43 Wi-Fi errors.
Other examples (e.g., sensor tests) work fine, so this issue seems limited to MJPEG streaming.

Steps to Reproduce:
1. Connect the Nicla Vision board and open the Arduino IDE.
2. Since Wi-Fi functionality is not enabled by default, flash the firmware using WiFiFirmwareUpdater (screenshot attached).
3. After this step, all Arduino Wi-Fi examples run successfully.
4. Open OpenMV IDE, flash the required firmware, and connect the board.
5. Run the MJPEG streaming example with valid Wi-Fi credentials.
6. The board connects to Wi-Fi and displays an IP address, but the webpage does not load.

Expected Behavior:

- The webpage should load correctly.
- The live MJPEG camera feed should be visible.

**Please complete the following information**
 - Arduino IDE 2.3.6
 - Arduino Mbed OS Nicla Board : 4.41
 - Board: Arduino Nicla Vision
 - OS: Windows
 - Open MV : 4.7.0

Screen shots: "https://drive.google.com/drive/folders/1ZY1QmEdSEmuMaRXb1_oefqj2LZ9kjoPl"

kwagyeman, 2025-08-25:
Hi, OpenMV IDE loads the correct firmware for the WiFi chipset itself. You do not need to use Arduino IDE in any way. Can you double-click on the reset button on the camera so that the onboard LED is pulsing? Then connect via OpenMV IDE, it will ask if you want to install the latest default firmware. Do so, and then the IDE will install the latest software for everything. You should be able to run the example then without any issues.

kwagyeman, 2025-08-25:
[image]

Latest firmware release works as expected.

closed 2025-08-26 by Ssumeetdubas

Ssumeetdubas, 2025-08-26:
please look on this ....

iabdalkader, 2025-08-26:
We don't offer any support for Arduino IDE or Arduino firmware. If you have any problem with our firmware (**OpenMV** firmware) or our IDE (**OpenMV** IDE) feel free to ask on the our forums, or open an issue if you think there's a bug somewhere. For any other Arduino problems, please ask on Arduino's forums.